riscv-software-src/riscv-perf-model #1: Add a branch predictor

klingaard, 2022-09-23:
Need a branch predictor to Fetch. Open to considerations on the type. Should be parameterized, but a common API would be beneficial to keep the model generic.

Resolutions of the branches will require feedback from Retire and/or Execute . Typicality this can be done with ports.

arupc, 2023-06-15:
There is an example branch prediction uarchitecture described here:  https://docs.boom-core.org/en/latest/sections/branch-prediction/index.html . This can serve as an example to implement in Olympia

venky528, 2023-07-10:
Unable to get a view of current status of this ticket. @arupc could you please brief about its current status.

light2802, 2023-08-11:
We currently have made a timing sheet for the branch predictor based on the BOOM architecture. The plan is to model and implement the branch predictor APIs based on this timing sheet and integrate it with Olympia.

[[1] Branch Predictor timing sheet](https://docs.google.com/spreadsheets/d/1NNue_sBiCrq0VSX1MiECl7uqAFufwIzgk8j7p1gVFro/edit?usp=sharing)

egieske, 2023-11-09:
What is the most efficient way to determine if a Fetched Inst is a branch? Should we add a member to Inst like Inst::isStoreInst() or isLoadStoreInst()?
Similarly, how should a branch predictor determine branch instruction type, e.g., call, return, unconditional, etc?
Also, how should a branch predictor monitor the execution status of branch instructions? Inst::getStatus()==COMPLETE seems possible, but then the predictor will have to scan a list of in-flight branch Inst's every cycle to determine when it can update predictor internal state which may be inefficient.

klingaard, 2023-11-09:
> What is the most efficient way to determine if a Fetched Inst is a branch?
I have an answer to that question here: https://github.com/riscv-software-src/riscv-perf-model/issues/104#issuecomment-1785229711

> Similarly, how should a branch predictor determine branch instruction type, e.g., call, return, unconditional, etc?

Mavis supports multiple instruction type identifiers.  You can see them here: https://github.com/sparcians/mavis/blob/4f3fef891f9ddc5c371c27500d02596f21ea6fc8/mavis/InstMetaData.h#L30

So, if you want to know if a branch is JALR, you can add this to Inst.hpp:
```
     bool isJALR() const { return mavis_info_->isInstType(mavis::OpcodeInfo::InstructionTypes::JALR); }
```

> Also, how should a branch predictor monitor the execution status of branch instructions? 

The predictor _should not monitor_ anything.  Doing so means you're no longer event driven.  😄  How to solve this: add a `sparta::DataOutPort<InstPtr>` port to the ExecutePipe.  When the ExecutePipe complete's the branch, it will write that branch to the port.  The predictor will include a `sparta::DataInPort<InstPtr>` that receives the completed branch and do its magic.  WARNING: you might get completed branches out of order.  😉

klingaard, 2023-11-09:
> What is the cost of adding sparta::DataOutPort port to the ExecutePipes...

None if you don't use it.  The ExecutePipe in the complete inst block can check if the instruction is a branch (in general) and send it on:
```
    if(ex_inst->isBranch()) {
        out_branch_complete_->send(ex_inst);
    }
```

The tradeoffs between an Data port and a Signal port (in this case) is use case.  Signal port says, "Some branch completed, somewhere."  The Data port says, "This specific branch completed."

egieske, 2023-11-09:
A SignalOutPort can be tied to multiple SignalInPorts (i.e., broadcast). Can multiple SignalOutPorts be tied to one SignalInPort with a wired-OR effect?

klingaard, 2023-11-09:
> A SignalOutPort can be tied to multiple SignalInPorts (i.e., broadcast). Can multiple SignalOutPorts be tied to one SignalInPort with a wired-OR effect?

Yep.  If not, let me know.  😉

egieske, 2023-12-01:
How can we determine if a branch instruction read from the trace is taken even if the branch is in the compact 'C' ISA form?
Except for the compact encoding case, we can check if the Inst is a branch (as above) && (target != (PC+4)).
Can we use something like:`bool isCompactISA() const { return mavis_info_->isInstType(mavis::OpcodeInfo::ISAExtension::C); }` and if TRUE check if (target != (PC+2)) ?
Defining bool isCompactISA() in Inst.hpp may also facilitate more accurate Fetch to Decode modeling where the hardware isn't limited to a fixed maximum amount of instructions as is currently in Olympia but is instead limited to a fixed maximum amount of bytes. For example, up to 32 bytes may be fetched and sent to Decode per cycle which could contain between 8x 4-byte and 16x 2-byte instructions.

danbone, 2023-12-02:
@egieske I think we have a mechanism provided already see isTaken()
https://github.com/danbone/riscv-perf-model/blob/0820c37c95068352d2ca7fe9cffb376c2e68fb2a/core/InstGenerator.cpp#L197

I think this comes from the STF reader. If using a different trace format then we'd have to implement it ourselves.

egieske, 2023-12-04:
Dan,
Thanks! I recalled that I had seen that, but when I did a search for isTaken and got nothing, then poked around mavis and saw nothing, I assumed my memory was mistaken and made my comment above.