riscv-software-src/riscv-perf-model #3: Make files more organized: each unit should have own directory

klingaard, 2022-09-23:
As the model grows in size/usefulness each unit should really be in its own directory coupled with specific unit testing.

jyrj, 2023-05-16:
Is this still up? I would like to work on this. I was going through an open source cycle accurate perf model and catch this. I think this issue will be a good start to understand the codebase.

klingaard, 2023-05-16:
You're more than welcomed to take this.  It does require the owner to create unit testing for each component, or at least the start of a unit test even if all it does is push instructions through.

For the CMake structure two suggested approaches:
1. Each component is built a its own library (.a extension) and linked in to the final `olympia` binary
2. A top-level CMakeLists.txt file that creates a `libcore.a` file that's linked into the final `olympia` binary
3. Other idea?