riscv-software-src/riscv-perf-model #4: Add unit testing framework and unit tests for each component

klingaard, 2022-09-23:
Each component (Fetch, Decode, etc) should have its own unit test.  To facilitate this, a testing framework should be written first and each unit using that.

knute-sifive, 2022-11-26:
First cut of a "common" set of source/sink units are made available in PR https://github.com/riscv-software-src/riscv-perf-model/pull/18

Next work to be done: generic simulator objects, although that might not be feasible.

knute-sifive, 2023-05-13:
I'm going to consider this work complete for now.  I set up a common source and sink unit [here](https://github.com/riscv-software-src/riscv-perf-model/tree/master/test/core/common) and Aaron was able to make use of it in his Rename branch (with a few tweaks).

closed 2023-05-13 by knute-sifive