riscv-software-src/riscv-perf-model #11: cmake error while trying to build

arupc, 2022-10-07:
Trying to build olympia from this repo and ran into this:

```
(sparta_dev) [master] src/riscv-perf-model/release % mkdir release; cd release
(sparta_dev) [master] src/riscv-perf-model/release % cmake .. -DCMAKE_BUILD_TYPE=Release -DSPARTA_BASE=/path/to/map/sparta
-- The CXX compiler identification is GNU 10.4.0
-- Detecting CXX compiler ABI info
...
...
CMake Error at CMakeLists.txt:23 (include):
  include could not find requested file:

    /users/achakraborty/mywork/src/riscv-perf-model/stf_lib/cmake/stf-config.cmake

CMake Error at CMakeLists.txt:92 (add_subdirectory):
  The source directory

    /users/achakraborty/mywork/src/riscv-perf-model/stf_lib

  does not contain a CMakeLists.txt file.
```

Checked that when I build using the previous olympia repo, cmake is successful. May be we need to copy one more file from old repo?

knute-sifive, 2022-10-07:
You need to actually tell `cmake` where your SPARTA install is...
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DSPARTA_BASE=/path/to/map/sparta # <-- fix that
```

knute-sifive, 2022-10-07:
Also, when you cloned, did you recursively clone `olympia`?  That will clone the stf library too.
```
git clone --recursive git@github.com:riscv-software-src/riscv-perf-model.git
```

arupc, 2022-10-07:
My bad, forgot to clone recursively. Building successfully. 

And I did specify the correct path to sparta. Changed the issue description to reflect that.

Thank you.

closed 2022-10-07 by arupc