riscv-software-src/riscv-perf-model #14: Add execution driven support

arupc-vmicro, 2022-10-19:
Description TBD.

VictorYing, 2023-08-17:
Hello, I'm looking to build an execution-driven RISC-V core performance model, to enable studies of multicore systems running interacting software threads, or studies evaluating effects of speculative execution on, for example, cache state. It looks like you've considered going execution-driven as a potential feature for Olympia. Is this still something on your roadmap? If so, what sort of goals you would be interested in working towards, and would be interested in accepting or collaborating on support for execution-driven simulation?

knute-sifive, 2023-08-17:
Hi Victor, there is a discussion about this here: https://github.com/riscv-software-src/riscv-perf-model/discussions/68

Olympia is only about a year old and the basics of the pipeline design is still being fleshed out.  Adding a functional back-end to the model is basically a wish list item until resources (volunteers) can be allocated to work on it.  You want to volunteer?  😄 

The effort will consist of the following plans:
1. Define/determine which functional back-end tool to use (Spike, Dramajo, something new?)
2. Determine the requirements to support EDM: speculative path support?  functional speculation execution/rewind support? etc
3. Design an API to basically generate traces on-the-fly and feed into Olympia OR
4. Design an API to actually execute instructions at the correct moment in Olympia (drop trace)
5. etc, etc

So many issues/design points to consider, esp if we want Olympia to support multicore (think cohesive behaviors).  But that's a different set of requirements.

VictorYing, 2023-08-17:
Thanks for the pointer to the discussion and the mention of Dramajo which I haven't looked into yet. I'm exploring options right now and might be able to allocate time to work on something down the line.

arupc, 2024-06-25:
@klingaard Can you please complete the description of the issue, per discussion in the weekly sync?  cc: @knute-mips 

Please feel free to unassign yourself after completing the description

VictorYing, 2025-11-04:
As discussed in Monday's SIG meeting, It seems activity on Dramajo has stopped., but Tenstorrent has recently open-sourced improvements to whisper to enable its use to drive an execution-driven perf model.  Specifically, it implements an API that allows an external agent to tell it what PCs to fetch, decode, and speculatively execute, and when to flush or commit their executions or make stores globally visible to other harts: https://github.com/tenstorrent/whisper/blob/master/PerfApi.hpp

Tenstorrent has successfully used this API to integrate whisper into our internal CPU perf models which can now succesfully show the impact of speculative path execution (e.g., speculative execution causing changes in the behavior of caches and prefetchers) and we intend to continue to maintain this API and whisper as a project released under a permissive open-source license.

I wonder if Olympia maintainers and perf model SIG members have any feedback on some of my thoughts on the potential project integrating Olympia and Tenstorrent's whisper:

- Is it fair to propose in an RVI mentorship program application that it is a desired outcome to merge working code into Olympia's master branch with goals of maintaining the integration in the future as a reference design for execution-driven simulation? I assume there should be configuration to disable the use of whisper for users who want to run traces produced by any other ISS (e.g., in SPF format).

- The integration must be demonstrated on public test workloads-- which ones? If we actually get speculative path execution working, ideally we should have some hard-to-predict branches. I'm not sure if Dhrystone qualifies-- IIRC, a sufficiently advanced branch predictor will achieve fairly low MPKI in that workload. Would microbenchmarks suffice? I guess Olympia has limited support for system code-- do we focus on bare-metal execution (unprivileged code)? Internally, Tenstorrent runs full-system Linux emulation, but I guess sticking with bare-metal lowers the requirements and makes the project more doable for an intern.

- I think improving Olympia's branch prediction is probably out of scope for the intern who works on execution-driven simulation, but the quality of the predictor needs to be good enough to make results showing the impact of executing speculative paths in the shadow of branch mispredicts interesting. Does this document accurately reflect the current state of Olympia's branch prediction in Olympia's default configuration? https://github.com/riscv-software-src/riscv-perf-model/blob/master/docs/bpu_design_document/BPUDesignDoc.adoc#basepredictor If so, I think this GShare predictor would be good enough that it would be an interesting outcome to show some wrong-path execution in the shadow of conditional branch direction mispredicts. If not, I wonder if other SIG members have plans to seek interns/mentees who can work on branch prediction?

  - Does the branch prediction depend on having "predecode" information such as whether or not a given PC is a conditional branch? If the answer is yes that is okay, it means the initial project is likely to need to ask the instruction set simulator to decode instructions at an early stage around when branch prediction is initiated, not later during the perf model's actual instruction fetch/decode stages.

- Is it accurate to say that Olympia does not currently have a fully working pipeline flush mechanism that is in active use? In that case, is getting flushes working across all the units an area where an intern can get guidance from other SIG members who have been more involved in prior Olympia development?

klingaard, 2025-11-06:
Loving the discussion... my thoughts (and requirements 😁).

Olympia is pretty much ready to go for EDM (Execution Driven Mode).  The simulator does have a well tested and working flush mechanism.   

Regardless of the state of branch prediction, having EDM support is powerful enabling users of the model to run binaries directly on the simulator instead of the extra step of trace generation (although there are MANY benefits of trace generation as well).

To facilitate an intern project adding EDM to Olympia, I would impose the following requirements:

### Olympia Pipeline Requirements

1. The Olympia project must remain agnostic to the functional driver of EDM.  Specifically, a user of Olympia should be able to provide their own functional model backend to drive EDM
2. The frontend of the simulator (Fetch) must remain agnostic to how the instructions are supplied to the simulator.  This is currently being done via the InstGenerator class.  A derivative of this class can be created that interacts with the functional model backend
3. Olympia backend must remain agnostic to the fact that it is running in EDM.  Outside of possible instruction latency adjustments due to data-dependent behaviors, the middle machine (for example) will not know nor need to know the mode of operation
4. Branch prediction can be aware of the mode of operation and steer the frontend down other paths of execution, like speculative paths
5. Trace and json flows must be preserved
6. Olympia must be the driver of simulation and the functional model be the slave.  The functional model will not drive simulation
7. Add config-time `cmake` rules to add EDM support.  The user must explicitly clone the functional models to be used (i.e. Whisper, Spike, Dromajo, etc must not be a submodule)

### InstGenerator for EDM Requirements

1. A new derivative of the InstGenerator must be provided that provides the same look-and-feel as the trace/json iteration approach
2. A generic API for EDM (virtual interface) must be developed.  This API can follow the pattern of the current Whisper API, but must not use the data types associated with the Whisper API
3. An implementation of this API can and will use the Whisper simulator 
4. Olympia tracks iterators with each instruction to know where to "rewind" to upon a flush.  The EDM interface must adhere to that design pattern

This is my first cut on this project.  I propose a schedule for the intern project be:

1. Understand the problem and understand the existing flow
2. Using the Whisper API, design a generic API for EDM and the functionality required to implement it
3. Add testing for the generic API
4. Integrate this API into an InstGenerator instance (EDMInstGenerator?)
5. Create a `cmake` configuration that enables/disables EDM support
6. Bring in Whisper, build a proxy, and connect to Olympia