riscv-software-src/riscv-perf-model #15: Add support for RISCV V1.0 vector extension

arupc-vmicro, 2022-10-19:
* Add ISA support to Mavis
* Add ISA Olympia extension (json files)

AaronGChan, 2023-08-25:
For this issue, what are the next steps for it? I see that Mavis has vector ISA support (https://github.com/sparcians/mavis/blob/main/json/isa_rv64v.json)

knute-sifive, 2023-08-28:
You'll need to create an Olympia json file that parallel's the Mavis one and place it here:

https://github.com/riscv-software-src/riscv-perf-model/tree/master/arches/isa_json

Then, I suggest you write a handful of json files that test the vector addition.  Make sure vector load/stores go to the load/store unit and arithmetic ones go to newly added vector units.  

You might need to add a vector scoreboard.

This should be a good start.

gglangg, 2023-10-06:
@AaronGChan Hi! I'd like to ask how the current progress is going. Have you encountered any problems?

AaronGChan, 2023-10-06:
> @AaronGChan Hi! I'd like to ask how the current progress is going. Have you encountered any problems?

Hi @gglangg,
Currently still working through the uarch design of a VPU to model before starting the implementations. I have a [discussion](https://github.com/riscv-software-src/riscv-perf-model/discussions/89) up currently, and I will be adding current design/considerations for out-of-order. We've been discussing in the SIG between having an in-order vs out-of-order VPU, with the in-order being much simpler to implement. I've been busy with school, so haven't had too much time to work through this yet, but trying to allocate more time.

Some design points/assumptions we were planning on making to make implementation easier for now:
Assume always that VL=VLMAX
Ignore masks

sequencer, 2023-10-06:
FYI, https://github.com/chipsalliance/t1 is also another lane based implementation for RVV

AaronGChan, 2024-06-17:
Current implementation will incorporate a vector unit as an execution unit. The design is thus a unified system, where both vector and scalar instructions pass through most stages together, with a separate vector issuequeue for vector instructions. Documentation for current design can be found at https://github.com/riscv-software-src/riscv-perf-model/discussions/89.

Summary of Work that needs to be Done for Vector Support:
- [] VSET/basic vector-vector operations (In-progress)
- [] Complex Vector instructions (Gather, Scatter)
- [] Vector Load/Store
- [] Vector tracing support in Dromajo, identify workloads #173 
- [] Mask Support

Future Work:
- [] Vector Coprocessor Design
- [] Split VRF, lane design
- [] VSET Predictor