riscv-software-src/riscv-perf-model #16: Fix flushing to rewind Fetch back to flush point

knute-sifive, 2022-11-23:
For STF reading (and possibly JSON), when the core is flushed, Fetch needs to refetch instructions from the flushed point.  Currently it just flushes and keeps going.

dingiso, 2022-12-31:
After attending the perf-model meeting, I got the point that the difficulties are the iterator always goes forward and the STF cannot be read using the index. 

So I think we can store the iterator for every branch instruction ( maybe store it in Inst class ), assign it to `next_it_` when flushing to rewind Fetch back to the flush point

klingaard, 2022-12-31:
> So I think we can store the iterator for every branch instruction ( maybe store it in Inst class ), assign it to `next_it_` when flushing to rewind Fetch back to the flush point

Yep, you can definitely do that.  Or, since it's copyable, store the STF iterator to the instruction itself.  Then we can support the two types of flushes:

1. Inclusive: The instruction should be included in the flush (think about loads/stores that miss in the TLB)
2. Redirecting: The instruction should _not_ be included in the flush, but fetching starts at the _next_ instruction (think mispredicted branches, RFIs, etc)

If the instruction had the iterator to itself, this can be used to start fetching from either the flushing instruction OR the next instruction.  Remember, you can `++` the iterator to get to the next instruction.

lata-11, 2023-05-06:
Hey! I am participating in LFX with this project. Can I work on this issue?

klingaard, 2023-05-07:
Absolutely!  Feel free to assign to yourself and let us know (either in this issue or in a discussion post) how you'll be tackling it.  Thanks!