riscv-software-src/riscv-perf-model #20: Compilation error on Apple Silicon box

aka2en, 2022-11-26:
```
/Users/zen/Sparta/riscv-perf-model/core/Inst.hpp:216:31: error: must specify at least one argument for '...' parameter of variadic macro [-Werror,-Wgnu-zero-variadic-macro-arguments]
                              SPARTA_ADDPAIR("complete",  &Inst::getCompletedStatus),                                                   
                              ^                                                                                                         
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:38:29: note: expanded from macro 'SPARTA_ADDPAIR'                      
#define SPARTA_ADDPAIR(...) _ADDPAIR_UTIL(__VA_ARGS__)                                                                                  
                            ^                                                                                                           
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:35:75: note: expanded from macro '_ADDPAIR_UTIL'                       
#define _ADDPAIR_UTIL(p_1, ...) addPair(p_1, std::forward<Args>(args)..., _ADDPAIR_RESOLVE(__VA_ARGS__)                                 
                                                                          ^                                                             
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:32:77: note: expanded from macro '_ADDPAIR_RESOLVE'                    
#define _ADDPAIR_RESOLVE(...) GET_ARGS(__VA_ARGS__, _RESOLVED_2, _RESOLVED_1)(__VA_ARGS__)                                              
                                                                            ^                                                           
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:29:9: note: macro 'GET_ARGS' defined here                              
#define GET_ARGS(_1, _2, FCN_NAME, ...) FCN_NAME                                                                                        
        ^                                                                                                                               
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.cpp:4:                   
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.hpp:15:     
In file included from /Users/zen/Sparta/riscv-perf-model/core/CoreTypes.hpp:6:                            
/Users/zen/Sparta/riscv-perf-model/core/Inst.hpp:217:31: error: must specify at least one argument for '...' parameter of variadic macro [-Werror,-Wgnu-zero-variadic-macro-arguments]
                              SPARTA_ADDPAIR("unit",      &Inst::getUnit),
                              ^                                                                                                         
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:38:29: note: expanded from macro 'SPARTA_ADDPAIR'
#define SPARTA_ADDPAIR(...) _ADDPAIR_UTIL(__VA_ARGS__)                                                                                  
                            ^                                                                                                           
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:35:75: note: expanded from macro '_ADDPAIR_UTIL'
#define _ADDPAIR_UTIL(p_1, ...) addPair(p_1, std::forward<Args>(args)..., _ADDPAIR_RESOLVE(__VA_ARGS__)
                                                                          ^
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:32:77: note: expanded from macro '_ADDPAIR_RESOLVE'
#define _ADDPAIR_RESOLVE(...) GET_ARGS(__VA_ARGS__, _RESOLVED_2, _RESOLVED_1)(__VA_ARGS__)
                                                                            ^
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:29:9: note: macro 'GET_ARGS' defined here       
#define GET_ARGS(_1, _2, FCN_NAME, ...) FCN_NAME                                                                                        
        ^                                                                                                                               
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.cpp:4:                                             
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.hpp:15:                  
In file included from /Users/zen/Sparta/riscv-perf-model/core/CoreTypes.hpp:6:
/Users/zen/Sparta/riscv-perf-model/core/Inst.hpp:218:31: error: must specify at least one argument for '...' parameter of variadic macro [-Werror,-Wgnu-zero-variadic-macro-arguments]
                              SPARTA_ADDPAIR("latency",   &Inst::getExecuteTime),
                              ^
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:38:29: note: expanded from macro 'SPARTA_ADDPAIR'
#define SPARTA_ADDPAIR(...) _ADDPAIR_UTIL(__VA_ARGS__)                                                                                  
                            ^                                                                                                           
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:35:75: note: expanded from macro '_ADDPAIR_UTIL'                       
#define _ADDPAIR_UTIL(p_1, ...) addPair(p_1, std::forward<Args>(args)..., _ADDPAIR_RESOLVE(__VA_ARGS__)
                                                                          ^   
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:32:77: note: expanded from macro '_ADDPAIR_RESOLVE'
#define _ADDPAIR_RESOLVE(...) GET_ARGS(__VA_ARGS__, _RESOLVED_2, _RESOLVED_1)(__VA_ARGS__)                        
                                                                            ^
/Users/zen/Sparta/map/sparta/sparta/pairs/RegisterPairsMacro.hpp:29:9: note: macro 'GET_ARGS' defined here
#define GET_ARGS(_1, _2, FCN_NAME, ...) FCN_NAME                                                                                        
        ^                                                                                                                               
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.cpp:4:    
In file included from /Users/zen/Sparta/riscv-perf-model/mss/BIU.hpp:15:                                            
In file included from /Users/zen/Sparta/riscv-perf-model/core/CoreTypes.hpp:6:            
```

knute-sifive, 2022-11-26:
Hmmm... thought this was fixed in my last push to map.  Have you pulled since last week?

aka2en, 2022-11-27:
No, I'll give it a try. Thanks!

aka2en, 2022-11-27:
Works well by the latest map.

closed 2022-11-27 by aka2en