riscv-software-src/riscv-perf-model #28: Error trying to build olympia

arupc-vmicro, 2022-12-06:
Running to the following error trying to build olympia

```
% cmake .. -DCMAKE_BUILD_TYPE=Release -DSPARTA_BASE=<path/to/>/map/sparta
...
-- Build files have been written to: /users/achakraborty/mywork/src/riscv-perf-model/release
% make olympia
...
[ 44%] Building CXX object core/CMakeFiles/core.dir/Fetch.cpp.o
/users/achakraborty/mywork/src/riscv-perf-model/core/Fetch.cpp: In member function 'void olympia::Fetch::fetchInstruction_()':
/users/achakraborty/mywork/src/riscv-perf-model/core/Fetch.cpp:76:52: error: conversion from 'std::size_t' {aka 'long unsigned int'} to
 'uint32_t' {aka 'unsigned int'} may change value [-Werror=conversion]
   76 |         credits_inst_queue_ -= insts_to_send->size();
```

klingaard, 2022-12-06:
Hey Arup, feel free to fix this by adding a cast.

arupc-vmicro, 2022-12-07:
fixed via #29