riscv-software-src/riscv-perf-model #32: cmake fails complaining about relative paths found in include directories

jerinjoy, 2023-01-12:
I'm not sure if this is the right forum to discuss this. I built sparta on my Ubuntu system and ran cmake in the Olympia directory which results in cmake complaining about relative paths in the include directories for zstd.

```
❯ cmake .. -DCMAKE_BUILD_TYPE=Release -DSPARTA_BASE=/scratch/joy/workspace/map/sparta
..
..
zstd_LIBRARIES (ADVANCED)
    linked by target "olympia" in directory /home/joy/workspace/riscv-perf-model
    linked by target "Dispatch_test" in directory /home/joy/workspace/riscv-perf-model/test/core/dispatch

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of "olympia":

    "zstd_INCLUDE_DIRS-NOTFOUND"

CMake Error in core/CMakeLists.txt:
  Found relative path while evaluating include directories of "core":

    "zstd_INCLUDE_DIRS-NOTFOUND"

CMake Error in mss/CMakeLists.txt:
  Found relative path while evaluating include directories of "mss":

    "zstd_INCLUDE_DIRS-NOTFOUND"
```

danbone, 2023-01-12:
I think its complaining that you have installed zstd

https://github.com/facebook/zstd

klingaard, 2023-01-13:
Correct, olympia uses STF library which requires zstd.  The requirement is
documented on the STF library README
<std::unique_ptr<sparta::Parameter<uint32_t>> num_iex_pipes_;
std::unique_ptr<sparta::Parameter<uint32_t>> num_fex_pipes_;
std::unique_ptr<sparta::Parameter<uint32_t>> num_vex_pipes_;>, but not
obvious for the olympia builder.  I'll update the olympia README.  Thanks
for bringing this to our attention.

closed 2023-01-13 by knute-sifive

jerinjoy, 2023-01-13:
thanks, that worked.