riscv-software-src/riscv-perf-model #35: Boost 1.81 not working (on macOS) - non-default boost location not supported by CMakeLists

peter-d, 2023-02-03:
Hi,

I've had some issues for Olympia to build on macOs (M1, running Ventura 13.1), not sure if the issue is present in other systems: default boost installed with ```brew install boost``` gives v1.81, which has issues in Phoenix which cause it to fail with duplicate symbols when linking (https://github.com/boostorg/phoenix/issues/111). (also Sparta build fails on this)

Just for other people running into this, here's my journey on this issue:

First of all: not to worry, I've also got boost v1.76 via homebrew, but it's not in a default location. So I had to build Sparta, Olympia and so also sta_lib with a non default boost location.

Then it got a bit complicated:
- Sparta build needs ```-DBoost_DIR=/opt/homebrew/Cellar/boost@1.76/1.76.0_3/lib/cmake/Boost-1.76.0``` to point it to the cake folder for the right boost version. (it does not, for some reason, use the cmake default included files to find boost)
- Olympia includes ```sparta-config.cmake``` so needs the same command line option for make to find Boost.
- But Olympia includes sty_lib in the build tree as a submodule, which does use the cmake default scripts, and they need another define on the command line: ```-DBOOST_ROOT=/opt/homebrew/Cellar/boost@1.76/1.76.0_3```

Which makes for a cmake command that gets a bit verbose:
```
cmake -S . -B Release -DCMAKE_BUILD_TYPE=Release -DSPARTA_BASE=$HOME/src/map/sparta \
   -DBOOST_ROOT=/opt/homebrew/Cellar/boost@1.76/1.76.0_3 \
   -DBoost_DIR=/opt/homebrew/Cellar/boost@1.76/1.76.0_3/lib/cmake/Boost-1.76.0
```

Which then passes cmake but fails at build, because boost is not added as an include for ```core``` and ```mss```.

Which I've fixed in the respective CMakeLists.txt, PR incoming soon.

But more generally: it would be nice to harmonise the boost config across Sparta, Olympia and stf_lib, so only 1 command line arg is needed to use it if it's in a non-default place. This avoids the double command line args needed in cmake.

I am willing to file a PR against Sparta as well to harmonise it to the standard cmake way of finding boost, but the comments in the CMakeLists.txt there seem to hint the custom way is there for (historic) reasons.

peter-d, 2023-02-03:
According to the https://github.com/boostorg/phoenix/pull/112#issuecomment-1332586502 a workaround for the whole 1.81 issue is to do ```-DBOOST_PHOENIX_STL_TUPLE_H_``` which might have other issues. If we go this route, this would need to be added to Sparta and stf_lib...

knute-sifive, 2023-02-03:
Hi Peter, totally agree.  The Boost `cmake` "finding policy" is a mess between the libraries and should be sync'ed.  Yes, please file a PR against sparta removing the APPLE constraints on Boost.  I'll do a little testing on the side and if all good, will approve and we can remove this nightmare.  

Thanks!

peter-d, 2023-02-04:
Hi Knute,

I'll let you know once I got it all lined up nicely in the PRs before we merge. 

I'd like to review and clean up the  boost include settings in the cmake too, in Sparta, it's added globally instead of specific targets, which makes you include boost when it might not be needed.

PR #42 (2023-05-08): Update boost deps for core and mss, track Sparta PR397
This fixes #35 and tracks the changes in Sparta [PR397](https://github.com/sparcians/map/pull/397)
commit 32fc79d5 (2023-05-08): Update boost deps for core and mss, track Sparta PR397 (#42)

This fixes #35 and tracks the changes in Sparta
[PR397(https://github.com/sparcians/map/pull/397), which are merges into Sparta main already. 

---------
closed 2023-05-08 by peter-d