riscv-software-src/riscv-perf-model #52: Enhance Olympia performance features

knute-sifive, 2023-06-06:
This GH issue is a starting point, or set of tasks, to add more functionality to Olympia based on what exists in the open-source community.  

Those tasks are:

- [ ] Find/list those open-source performance simulators in the industry that are similar to Olympia (OoO, super scalar models, not necessarily RISC-V based)
- [ ] For each simulator found, list what that simulator supports/has that Olympia lacks
- [ ] Add GH issues to this project so that discussion/work can be determined

Once GH issues are added, this GH issue can be considered complete.

Shubhf, 2024-08-10:
Hey @peter-d @h0lyalg0rithm Can I take up this issue? 
I am thinking that first we need to identify and analyze open source platform simulators which are similar to Olympia, pinpointing features they support that Olympia lacks. Based on the analysis, create actionable Github Issues for each missing feature to  guide further development and enhance Olympia's functionality.

peter-d, 2024-08-10:
Thank you for the initiative! I’m not following up closely on Olympia nor actively contributing at this moment, so I can’t really answer. Maybe worth checking with @klingaard as well, instead of me.