riscv-software-src/riscv-perf-model #55: Generate STF trace for Coremark using STF instrumented Dromajo

arupc, 2023-06-15:
Try out the steps to generate dhrystone stf trace following instructions from https://github.com/riscv-software-src/riscv-perf-model/blob/master/traces/README.md . This is the suggested order of operations: 
         - replicate the current instructions in the README using the specified Dromajo SHA (which is old now) and generate Dhrystone trace
         - repeat the same steps with latest Dromajo SHA to see if its still working
         - repeat the same steps as above to generate coremark trace
         -  look into possibilities of creating and maintaining a fork of Dromajo with STF generation capabilities instead of a patch.

jeffnye-gh, 2023-10-16:
Hi, 

Is there anyone working on this ? I can help. 

I got the latest dromajo patched and generating z/stf from linux and baremetal.

klingaard, 2023-10-16:
I just merged a PR (PR #67) last week that patches a newer Dromajo, but not the latest master.  I was able to produce an updated Dhrystone, but didn't try Coremark.  

Feel free to merge with latest mast of Olympia, update the patch and put in a PR.  Thanks!

jeffnye-gh, 2023-10-16:
my bad, I missed your recent commit. I was after the uart performance improvement in a version of dromajo that can also generate stf. I used to maintain a patched (slow) and 'head' (fast) version. 

A quick scan of the changes between april and aug (latest) does not show anything significant to me. I don't think it's worth the churn for another PR/merge right now.

knute-sifive, 2023-10-16:
No worries.  I don't like my fix for now -- restricting Dromajo's instruction quantum to 1 for EVERYTHING.  I think a smarter approach is to allow Dromajo to run as fast as possible until it hits a trace start maker.  Then, it goes into crawl mode.  If you have any ideas or patches that do that, would love to have 'em!

jeffnye-gh, 2023-10-24:
I have a branch that moves tracing generation into the interpreter and so keeps performance when not tracing. 

I need some guidance on how to proceed.  I created a branch on riscv-perf-model, push fails with permission denied. What's process do others use for this? 

console:
git push origin jeffnye.dromajo_stf_update
Username for 'https://github.com': jeffnye-gh
Password for 'https://jeffnye-gh@github.com':  <added my jeffnye-gh account token>
remote: Permission to riscv-software-src/riscv-perf-model.git denied to jeffnye-gh.

fatal: unable to access 'https://github.com/riscv-software-src/riscv-perf-model.git/': The requested URL returned error: 403

knute-sifive, 2023-10-24:
For pubic projects with many developers, it’s obviously not feasible to give everyone push access to a repo.

What folks do is fork the repo into their own private space, create branches and push to those branches to their private forks.  

To push your private branch to the main repo, you start by creating a new pull request (I believe in your private fork) and set pull downs to the proper source/destination locations.