riscv-software-src/riscv-perf-model #57: Improve Load/Store Unit

knute-sifive, 2023-06-26:
The load/store unit in the RISC-V Performance Model is a simple, single, in-order, pipeline ([sparta::Pipeline](https://sparcians.github.io/map/classsparta_1_1Pipeline.html)) with a fixed number of stages: MMU Lookup, Cache lookup, and completion.  

To enhance this unit, or rather to make it more extensible, the taker of this GH issue must understand the current limitations of the block.

There is no unit test for the load/store unit
- [ ] Add a unit test similar to Rename and Dispatch
- [ ] Add testing for various loads and stores; ensure timing looks sane for given parameters

The pipeline stages are hardcoded: https://github.com/riscv-software-src/riscv-perf-model/blob/f7c17dcb8b10f80650fa50cbc1affe31e3a0624f/core/LSU.hpp#L98-L104 with fixed latencies and length
- [ ] Fix 1: Remove the `enum` and replace with load/store parameters; verify current behavior is unchanged.  Test the

The MMU can only handle 1 outstanding miss
- [ ] Add support to allow the MMU to service multiple outstanding misses; consider merging multiple misses into 1 access

The cache can only handle 1 outstanding miss
- [ ] Add support to allow multiple cache misses; consider merging multiple misses to the same line into 1 miss

There is only support for 1 load/store pipeline
- [ ] Add support for _N_ load/store pipelines; this entails ensuring there are no collisions on the cache/mmu blocks between pipes (or add explicit support to allow that and assert when NOT requested)

Arbitration for load/store pipe is [simple](https://github.com/riscv-software-src/riscv-perf-model/blob/f7c17dcb8b10f80650fa50cbc1affe31e3a0624f/core/LSU.cpp#L567)
- [ ] Add support for a smarter arbitration: loads take priority over stores; consider age, etc

Separate loads from stores
- [ ] Stores are typically not important.  Add support to move them off to the side

Other ideas welcomed...

knute-sifive, 2023-07-07:
Additional improvements:
- [ ] Support merging of loads/stores that fall into the same cache line into existing/outstanding transactions
- [ ] Document existing behaviors and planned behaviors (on the [wiki](https://github.com/riscv-software-src/riscv-perf-model/wiki))
- [ ] Disconnect the LS from the L1 cache -- instead, have the LS communicate with a MSS-like unit that _contains_ the L1 (and/or L2, and/or L3, etc).  The MSS unit is documented in #13

danbone, 2024-01-08:
Any reason why #91 was closed? It'll allow the DCache to handle more than 1 miss at a time?

klingaard, 2024-01-08:
#91 was merged and adds _some_ of the requested support listed above.  More to come!

danbone, 2024-01-08:
I think there must have been an error in the merge or something, as none of the DCache changes* from #91 made it into #92 or into main?

*haven't checked other files.

klingaard, 2024-01-08:
Ugh, I misread the history of #91 -- you're right, they didn't get merged directly.  I _thought_ that #92 was supposed to include them.  @h0lyalg0rithm do you mind getting to the bottom of this?

h0lyalg0rithm, 2024-01-09:
Hi Knute
I am ooo at the moment. But I assumed Vineeth message you privately letting
you know that his PR is still not merged as it doesn't connect to the new
L2 cache.
I had a quick message with today and he mentioned that he is working on it.

I don't have access to a laptop atm, will look into it on Saturday.
Apologize for the delay