riscv-software-src/riscv-perf-model #61: Model cache replacement policies

arupc, 2023-06-30:
Develop replacement policies: random, LRU, TreePLRU using Sparta for various caches in Olympia.

Add a documentation on how to add more complicated replacement policies.

govardhnn, 2024-11-21:
Hi @arupc @klingaard, I would like to get started on Olympia by taking this up. 

I see that sparta has implementations of LRU and MRU, other than the TreePLRU used in Olympia.

To make the current replacement policy in Olympia configurable, I propose to:
Parameterize `setupL1Cache_` in the file `DCache.cpp` and the constructor in the file `ICache.cpp`  - with default being TreePLRU for backward compatibilty.

 I would then build olympia - for all the arches with all the new configurations,
 
 And run tests that check the accesses:
 1. When all lines are dirty/clean/invalid - pick the replacement way that the caches are configured to(LRU, MRU or TreePLRU) 
 2. Mix of dirty + clean, to pick the clean line
 3. Mix of invalid + dirty + clean to pick the invalid line
 
I shall also adhere to the coding guidelines and write an adoc for design documentation as specified by @jeffnye-gh 
This shall consist of implementation details, configuration options, performance implications and usage examples.

Is there anything else I should be doing - please let me know.

klingaard, 2024-11-21:
Sounds good!  Be happy to assign this to you.  Thank for you for volunteering.

govardhnn, 2024-11-24:
Thank you. I have started [discussion#231](https://github.com/riscv-software-src/riscv-perf-model/discussions/231) regarding this.