riscv-software-src/riscv-perf-model #81: Error building an STF Capable Dromajo

knute-sifive, 2023-08-23:
### Discussed in https://github.com/riscv-software-src/riscv-perf-model/discussions/80

<div type='discussions-op-text'>

<sup>Originally posted by **Nofal475** August 23, 2023</sup>
I am following the steps given to buid a STF-Capable dromajo, but for some reason it is failing 
I get a long error message when I run make, part of which is attached below
If I don't apply dromajo_stf_lib.patch I don't get this error 
Error:
 /home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/type_traits:910:52: error: static assertion failed: template argument must be a complete class or an unbounded array
  910 |       static_assert(std::__is_complete_or_unbounded(__type_identity<_Tp>{}),
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/type_traits:910:52: note: 'std::__is_complete_or_unbounded<__type_identity<stf::TraceInfoRecord> >((std::__type_identity<stf::TraceInfoRecord>(), std::__type_identity<stf::TraceInfoRecord>()))' evaluates to false
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/stl_construct.h: In instantiation of 'void std::_Destroy(_ForwardIterator, _ForwardIterator) [with _ForwardIterator = stf::TraceInfoRecord*]':
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/alloc_traits.h:850:15:   required from 'void std::_Destroy(_ForwardIterator, _ForwardIterator, allocator<_T2>&) [with _ForwardIterator = stf::TraceInfoRecord*; _Tp = stf::TraceInfoRecord]'
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/stl_vector.h:730:15:   required from 'std::vector<_Tp, _Alloc>::~vector() [with _Tp = stf::TraceInfoRecord; _Alloc = std::allocator<stf::TraceInfoRecord>]'
/home/nofal/Workspace/Student/map/riscv-perf-model/traces/stf_trace_gen/dromajo/stf_lib/stf-inc/stf_writer_base.hpp:24:11:   required from here
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/stl_construct.h:188:51: error: static assertion failed: value type is destructible
  188 |       static_assert(is_destructible<_Value_type>::value,
      |                                                   ^~~~~
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/stl_construct.h:188:51: note: 'std::integral_constant<bool, false>::value' evaluates to false
/home/nofal/miniconda3/envs/sparta/x86_64-conda-linux-gnu/include/c++/12.3.0/bits/stl_construct.h:195:25: error: invalid use of incomplete type 'std::iterator_traits<stf::TraceInfoRecord*>::value_type' {aka 'class stf::TraceInfoRecord'}
  195 |       std::_Destroy_aux<__has_trivial_destructor(_Value_type)>::
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nofal/Workspace/Student/map/riscv-perf-model/traces/stf_trace_gen/dromajo/stf_lib/stf-inc/stf_record_types.hpp:2055:11: note: forward declaration of 'std::iterator_traits<stf::TraceInfoRecord*>::value_type' {aka 'class stf::TraceInfoRecord'}
 2055 |     class TraceInfoRecord : public TypeAwareSTFRecord<TraceInfoRecord, descriptors::internal::Descriptor::STF_TRACE_INFO> {
      |           ^~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/dromajo.dir/build.make:76: CMakeFiles/dromajo.dir/src/dromajo.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:147: CMakeFiles/dromajo.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
</div>

knute-sifive, 2023-08-23:
@bdutro is able to reproduce and is looking at it.

knute-sifive, 2023-08-25:
So Brett reproduced a _different_ error that he fixed with an updated package (not related to this issue).  

We can't reproduce your issue.  Can you try with `clang` instead?

```
CC=clang CXX=clang++ cmake .. <args>
```

Nofal475, 2023-08-28:
I did all the steps again from installing stf_lib olympia and sparta and now it is working, i think there was a problem with my stf_lib build.
Thanks again for your help.

knute-sifive, 2023-08-28:
Good to hear.

closed 2023-08-28 by knute-sifive