riscv-software-src/riscv-perf-model #83: Linking CXX Executable issue

Krishnakumarmohanraj, 2023-08-30:
(sparta) [manicka.vinodhini@localhost release]$ make olympia
[  0%] Built target stf_git_version
[ 49%] Built target stf
[ 54%] Built target mss
[ 94%] Built target core
[ 96%] Linking CXX executable olympia
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_filesystem.so.1.78.0: error: undefined reference to '__cxa_throw_bad_array_new_length', version 'CXXABI_1.3.8'
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libyaml-cpp.so.0.7.0: error: undefined reference to 'std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()', version 'GLIBCXX_3.4.26'
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_serialization.so.1.78.0: error: undefined reference to 'std::uncaught_exceptions()', version 'GLIBCXX_3.4.22'

Installed server details:
OS: Centos 7
GCC version: 9.3.1(devtoolset-9)
Conda version: 23.7.3

h0lyalg0rithm, 2023-08-30:
@Krishnakumarmohanraj I think there was a mismatch in the gxx version that was used to build sparta vs the one used to build olympia.
Could you try out the following?
Try to build sparta using the gxx compiler part of the conda environment and then try to build olympia.

Hope this helps

knute-sifive, 2023-08-30:
Can you provide the output of `which gcc`?

Krishnakumarmohanraj, 2023-08-30:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> CAUTION: This email originated from outside the Organization. Do not open Link or Attachments unless you recognize the sender and know the content is safe.

Hi Mr.Knute Lingaard,

Details as below,

(sparta) ***@***.*** release]$ gcc --version
gcc (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(sparta) ***@***.*** release]$

Disclaimer

“This e-mail message (including any attachments) is intended to be received only by person(s) entitled to receive the confidential and/or legally privileged information it may contain. If you are not the intended recipient, you are notified that you have received this message by error, and any review, dissemination, distribution, storage or copying of the contents of this message is strictly prohibited. Please notify the sender immediately by email, and delete the message and its contents completely from your computer system and from all devices where it could have downloaded and all online Email inboxes, without disclosing it to anyone.”

knute-sifive, 2023-08-30:
Actually, my question was which gcc:
```
% which gcc
```

I want to see if it's the one in the sparta conda env.

Krishnakumarmohanraj, 2023-08-30:
Hi Mr. Knute Lingaard,

   Which gcc - point to devtoolset, i tried to find gcc in sparta, below is the result. Which one i need to point out

(sparta) ***@***.*** release]$ which gcc
/opt/rh/devtoolset-9/root/usr/bin/gcc

(sparta) ***@***.*** ~]$ cd miniconda3/
(sparta) ***@***.*** ~/miniconda3]$ find -name "gcc"
./pkgs/libgcc-devel_linux-64-12.3.0-h8bca6fd_0/lib/gcc
./pkgs/libstdcxx-devel_linux-64-12.3.0-h8bca6fd_0/lib/gcc
./pkgs/boost-cpp-1.78.0-h6582d0a_3/include/boost/mpl/aux_/preprocessed/gcc
./pkgs/gxx_impl_linux-64-12.3.0-he2b93b0_0/libexec/gcc
./pkgs/gcc_impl_linux-64-12.3.0-he2b93b0_0/lib/gcc
./pkgs/gcc_impl_linux-64-12.3.0-he2b93b0_0/libexec/gcc
./pkgs/gcc_impl_linux-64-12.3.0-he2b93b0_0/libexec/gcc/x86_64-conda-linux-gnu/12.3.0/gcc
./pkgs/gcc_impl_linux-64-12.3.0-he2b93b0_0/share/licenses/gcc
./envs/sparta/lib/gcc
./envs/sparta/share/licenses/gcc
./envs/sparta/include/boost/mpl/aux_/preprocessed/gcc
./envs/sparta/libexec/gcc
./envs/sparta/libexec/gcc/x86_64-conda-linux-gnu/12.3.0/gcc
(sparta) ***@***.*** ~/miniconda3]$

knute-sifive, 2023-08-30:
Ah.  The conda environment installs libraries and tools that all work together.  If you use a compiler _outside_ of the conda environment, there are no guarantees of compatibility.

Try this to get you started:
```
conda activate sparta
CC=clang CXX=clang++ cmake ..
make 
```

knute-sifive, 2023-08-30:
Make sure the clang you're using is in the conda environment.  If you prefer to use gcc, you must install gcc in the conda environment.

knute-sifive, 2023-08-30:
This should work:
```
% conda activate sparta
% conda install -c conda-forge gcc
% which gcc
/home/manicka.vinodhini/miniconda3/envs/sparta/bin/gcc
```

Krishnakumarmohanraj, 2023-08-31:
Hi Mr.Knute Lingaard,

     We installed gcc as per request, but still we facing same issue.

(sparta) ***@***.*** release]$ which gcc
/home/manicka.vinodhini/miniconda3/envs/sparta/bin/gcc
(sparta) ***@***.*** release]$ gcc --version
gcc (conda-forge gcc 12.3.0-0) 12.3.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(sparta) ***@***.*** release]$ make olympia
[  0%] Built target stf_git_version
[ 49%] Built target stf
[ 54%] Built target mss
[ 94%] Built target core
[ 96%] Linking CXX executable olympia
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_filesystem.so.1.78.0: error: undefined reference to '__cxa_throw_bad_array_new_length', version 'CXXABI_1.3.8'
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libyaml-cpp.so.0.7.0: error: undefined reference to 'std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()', version 'GLIBCXX_3.4.26'
/home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_serialization.so.1.78.0: error: undefined reference to 'std::uncaught_exceptions()', version 'GLIBCXX_3.4.22'

Thanks,
Krishna

knute-sifive, 2023-08-31:
So, this isn't an Olympia issue nor a conda one but possibly more of environment issue on your host machine.

The CXX ABI and glibc libraries that `boost` and `yaml` are dynamically linking are very, very old (2017).  

What is the output of this command?
```
ldd /home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_filesystem.so.1.78.0.so
```
My dynamic links show the use of libstd++ found in the conda environment, not local.

Krishnakumarmohanraj, 2023-08-31:
Hi Mr. Knute Lingaard,

Details as below,

(sparta) ***@***.*** release]$ ldd /home/manicka.vinodhini/miniconda3/envs/sparta/lib/libboost_filesystem.so.1.78.0
      linux-vdso.so.1 =>  (0x00007ffd86533000)
      libboost_atomic.so.1.78.0 => /home/manicka.vinodhini/miniconda3/envs/sparta/lib/./libboost_atomic.so.1.78.0 (0x00007facd7af0000)
      librt.so.1 => /lib64/librt.so.1 (0x00007facd76f1000)
      libstdc++.so.6 => /home/manicka.vinodhini/miniconda3/envs/sparta/lib/./libstdc++.so.6 (0x00007facd750e000)
      libm.so.6 => /lib64/libm.so.6 (0x00007facd720c000)
      libgcc_s.so.1 => /home/manicka.vinodhini/miniconda3/envs/sparta/lib/./libgcc_s.so.1 (0x00007facd7ab1000)
      libpthread.so.0 => /lib64/libpthread.so.0 (0x00007facd6ff0000)
      libc.so.6 => /lib64/libc.so.6 (0x00007facd6c22000)
      /lib64/ld-linux-x86-64.so.2 (0x00007facd78f9000)
(sparta) ***@***.*** release]$

Thanks,
Krishna

knute-sifive, 2023-08-31:
Ok.  Looks reasonable.  What's the output of this (yes, you are executing the `.so` -- a neat little trick most people don't know):
```
(sparta) ***@***.*** release]$ /lib64/libc.so.6 
```
This is my output:

```
GNU C Library (Ubuntu GLIBC 2.31-0ubuntu9.7) stable release version 2.31.
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 9.3.0.
libc ABIs: UNIQUE IFUNC ABSOLUTE
For bug reporting instructions, please see:
<https://bugs.launchpad.net/ubuntu/+source/glibc/+bugs>.
```

Krishnakumarmohanraj, 2023-08-31:
Hi Mr. Knute Lingaard,

 Output as below,

(sparta) ***@***.*** release]$ /lib64/libc.so.6
GNU C Library (GNU libc) stable release version 2.17, by Roland McGrath et al.
Copyright (C) 2012 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 4.8.5 20150623 (Red Hat 4.8.5-44).
Compiled on a Linux 3.10.0 system on 2022-05-18.
Available extensions:
      The C stubs add-on version 2.1.2.
      crypt add-on version 2.1 by Michael Glad and others
      GNU Libidn by Simon Josefsson
      Native POSIX Threads Library by Ulrich Drepper et al
      BIND-8.2.3-T5B
      RT using linux kernel aio
libc ABIs: UNIQUE IFUNC
For bug reporting instructions, please see:
<http://www.gnu.org/software/libc/bugs.html>.
(sparta) ***@***.*** release]$ cat /etc/redh
redhat-lsb/     redhat-release@
(sparta) ***@***.*** release]$ cat /etc/redhat-release
CentOS Linux release 7.9.2009 (Core)
(sparta) ***@***.*** release]$

Thanks,
Krishna

<https://bugs.launchpad.net/ubuntu/+source/glibc/+bugs>.

knute-sifive, 2023-08-31:
Yeah, that's your problem.  That `libc` is SUPER old.  Is there any way you update your system?  Maybe install a libc in conda?  Not sure how to do that -- you'll have to consult the internet.  😉

Krishnakumarmohanraj, 2023-08-31:
Hi Mr. Knute Lingaard,

 But this is default gcc version in Centos 7.9, whether it is works in centos 9 OS version or riscv-perl-model only works in ubuntu?

  Meanwhile, I will check in internet whether any option is available to update libc in conda

Thanks,
Krishna

knute-sifive, 2023-08-31:
Don't know much about Centos outside of RH acquiring them. But the `libc` that's with that distro was compiled with a REALLY old `gcc` from a really old RH distro:
```
Compiled by GNU CC version 4.8.5 20150623 (Red Hat 4.8.5-44).
```
Long story short -- unless you rebuild EVERYTHING that sparta and olympia depends on directly for your system, it just won't link with the out-of-box libraries in that distribution.  If you can get it working within `conda` that'd be best, but I don't think we should add/ensure support for systems that are significantly old.

Feel free to continue asking questions here, but I'm going to close this issue.  I'm not sure there's anything to fix on our side...

closed 2023-08-31 by knute-sifive