riscv-software-src/riscv-perf-model #93: Float to Integer instructions cause assertion: "Got an I2F instruction in an ExecutionPipe that does not source the integer RF"

danbone, 2023-10-05:
All of the float to integer instructions cause this assertion to fire within the execute pipe:

https://github.com/riscv-software-src/riscv-perf-model/blob/d74549339c2d70adae5b3fdd706bc6c44159f4b5/core/ExecutePipe.cpp#L148-L150

If the intention is that these instructions use the i2f target pipe (not a new f2i type), then I think the code just needs to be changed to handle f2i, and the assertion removed.

Test trace:
```
[
    {
        "mnemonic": "fcvt.w.s",
        "rs1": 1,
        "rs2": 2,
        "rd": 3
    }
]
```

knute-sifive, 2023-10-05:
Yep, appears to not have been tested.  Thanks for letting us know.

Better to support an `f2i` target to allow for future pipe assignments (maybe f2i is handled by FP pipes and `i2f` is handled by IEX pipes).

A tweak to your JSON input:
```
[
    {
        "mnemonic": "fcvt.w.s",
        "fs1": 1,
        "rd": 3
    }
]
```
`frm` is ignored...

Got a fix I'm testing now.

knute-sifive, 2023-10-05:
PR up.  Please take a look.

closed 2023-10-06 by knute-sifive