riscv-software-src/riscv-perf-model #96: Central allocator class for all allocations

knute-sifive, 2023-10-06:
One of the latest branches of Olympia is seg-faulting due to memory allocators being destroyed before units in simulation.  

When a unit in simulation is torn down, it will destroy internal members like queues, buffers, etc which might still be holding onto objects that were allocated using an Allocator.  Need to ensure the Allocators outlive the simulation components.

Ironically this was just asked in Map/Sparta: https://github.com/sparcians/map/discussions/445#discussioncomment-7166401

egieske, 2023-10-09:
Might this or something similar be the cause of a warning we are seeing as we try to add a basic L2 unit that contains a pipeline and a miss buffer between the LSU & BIU units? We've verified that the MemAccessInfoPtr references are incrementing and decrementing properly as the DCache miss moves through the L2 unit. The sim runs but gives this message during teardown:
WARNING: Seems that not all of the blocks made it back.  
'sparta::SpartaSharedPointerAllocator<PointerT>::~SpartaSharedPointerAllocator() [with PointerT = olympia::LSU::MemoryAccessInfo]'
Allocated: 2
Returned: 0

knute-sifive, 2023-10-09:
It is related and you might see this warning disappear after you pull latest master of Olympia.

But, to answer your question differently, that warning can happen when Retire ceases simulation (it's told a given instruction is the last), and there are still transactions in the MSS (related to old, but now long-gone instructions OR instructions that are not required to retire [simulation terminated via `-i` option]).   

The way Olympia ceases simulation:

1. Retire acknowledges the instruction limit: `-i inst_cnt` option
2. The last instruction of the trace is retired
3. No more events are scheduled on the Sparta Scheduler (this is most likely an error on the modeler's part)

This simulation=stopping methodology has always been a topic of conversation: Do you want ALL transactions to complete before simulation end?  Or do you no longer care about outstanding MSS transactions once the last instruction retires?

If you want the _entire simulator_ to drain of all transactions, then you simply prevent Retire from stopping simulation: https://github.com/riscv-software-src/riscv-perf-model/blob/0c98a64081f9671ef482f324827cf91dbb5c681c/core/ROB.cpp#L136
and hopefully the third bullet above stops simulation for you.  But that can be dangerous -- what if your model has a bug in it and is no longer scheduling events as it should?   The way to ensure that doesn't happen: implement https://github.com/riscv-software-src/riscv-perf-model/blob/0c98a64081f9671ef482f324827cf91dbb5c681c/core/LSU.hpp#L396 in your block and _make sure_ your queues/buffers are fully drained.

knute-sifive, 2023-10-14:
Fixed in PR #97

closed 2023-10-14 by knute-sifive