riscv-software-src/riscv-perf-model #109: Adding new sparta unit : L2Cache

Kunal-Buch, 2023-10-31:
- Create a new sparta unit for L2Cache that received requests from IL1 and DL1 and sends requests out to BIU for the misses in L2Cache.
- L2Cache can handle multiple outstanding/pending misses and supports out-of-order returns from BIU.
- Separate ports for acks/credits and responses for DL1, IL1 and BIU.

knute-sifive, 2023-11-01:
An example of CycleCounter/stall counting: https://github.com/riscv-software-src/riscv-perf-model/blob/82504f6f68374d4b335d9de3d5d3890fe55e9838/core/Dispatch.hpp#L132

Presentation on Sparta, including example of how to handle multiple inputs and choose one output:
https://docs.google.com/presentation/d/e/2PACX-1vS1BWtVv0x3qXKQWAeECe2gsF9cMG3Zp2HnXJw52grCAcl21lv3a9pLW6J0lZ32e5DWdZkFyUMcE_AI/pub?start=false&loop=false&delayms=3000

Move forward to "Ordered Scheduling"

Kunal-Buch, 2023-11-07:
Thanks Knute! :)

I have updated the model for the backpressure handling when the buffers are full.
I also have the make regress clean with the couple of tests added for the L2Cache.

Do I need permissions to create a remote branch and create a PR?

knute-sifive, 2023-11-07:
You can open a PR [here](https://github.com/riscv-software-src/riscv-perf-model/compare) that references your fork of Olympia.  Once done, we will receive a note to enable CI and then we'll look it over.  Thanks for contributing!

zhynsb, 2024-05-14:
Hi, guys

Could you write a README to run tests with L1+L2 system?

Thanks :)