riscv-software-src/riscv-perf-model #122: Not able to trace with Dromajo

shivampotdar, 2023-11-19:
Hi

I am following the instructions in https://github.com/riscv-software-src/riscv-perf-model/blob/master/traces/README.md for trying tracing with new workloads.

The steps do not work out of the box on latest checkout with Ubuntu 22.04. Particularly there were three errors in building buildroot. 
1. SIGSTKSZ not defined which is similar to that described in https://github.com/openwrt/openwrt/issues/9055 and could be resolved by applying patch from  https://launchpad.net/ubuntu/+archive/primary/+sourcefiles/m4/1.4.18-5ubuntu1/m4_1.4.18-5ubuntu1.debian.tar.xz
2. _STAT_VER not defined, that was fixed from the diff for libfakeroot.c here - https://salsa.debian.org/clint/fakeroot/-/merge_requests/10/diffs#6c4c023cdb2bb2d22a70b89b3dca80920ac0dd79
3. mknod operation not permitted in the cpio.mk step. The only fix I could find is to edit traces/stf_trace_gen/dromajo/run/buildroot-2020.05.1/fs/cpio/cpio.mk and add "sudo" before the mknod line. 

After these fixes I could perform all the build steps in the README, but when running Dromajo it fails with - 

```
[    0.594772] Freeing unused kernel memory: 232K
[    0.598231] Run /init as init process
mount: you must be root
mount: you must be root
mount: you must be root
mount: you must be root
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
hostname: sethostname: Operation not permitted
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Starting network: ip: RTNETLINK answers: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
```

This does not go away by running dromajo with root, nor I can perform the complete build under root as buildroot make infra complains that it is unsafe to do so. Trying this on a different machine did not help either.
I am stuck here and any suggestions to get past the issue would be much appreciated.

Thanks!

Full OpenSBI log if that helps -

```
sr_read: invalid CSR=0x30a
csr_read: invalid CSR=0xda0
csr_read: invalid CSR=0xfb0

OpenSBI v1.3-111-gdc0bb19
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : ucbbar,dromajo-bare
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x80000000
Firmware Size             : 194 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 66 KB
Firmware Heap Offset      : 0x28000
Firmware Heap Size        : 34 KB (total), 2 KB (reserved), 9 KB (used), 22 KB (free)
[... 142 lines cut ...]
[    0.581633] usbhid: USB HID core driver
[    0.584444] NET: Registered protocol family 10
[    0.586799] Segment Routing with IPv6
[    0.586970] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    0.588789] NET: Registered protocol family 17
[    0.589509] 9pnet: Installing 9P2000 support
[    0.589716] Key type dns_resolver registered
[    0.594772] Freeing unused kernel memory: 232K
[    0.598231] Run /init as init process
mount: you must be root
mount: you must be root
mount: you must be root
mount: you must be root
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
can't open /dev/null: No such file or directory
hostname: sethostname: Operation not permitted
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Starting network: ip: RTNETLINK answers: Operation not permitted
ip: SIOCSIFFLAGS: Operation not permitted
FAIL
```

shivampotdar, 2023-11-19:
Update - 
Turns out this happens only on virtual machines. I tried both WSL and GCP instances with similar outcome. Hence the instructions are okay, sorry about pointing that out.
Nonetheless, if someone knows how to get this working on WSL that would be great!

klingaard, 2023-11-20:
Curious, do you see the same issue _without_ the patch?  If so, this is probably something worth asking on the Dromajo project.

klingaard, 2023-12-13:
Closing this as there's no follow up.

closed 2023-12-13 by klingaard