riscv-software-src/riscv-perf-model #158: Cannot compile with Profile build type

danbone, 2024-03-05:
I'm trying to compile with CMAKE_BUILD_TYPE=Profile but I'm getting the following error:
```
/home/dan/anaconda3/envs/sparta/bin/../lib/gcc/x86_64-conda-linux-gnu/12.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: /home/dan/anaconda3/envs/sparta/bin/../x86_64-conda-linux-gnu/sysroot/usr/lib/../lib/gcrt1.o: relocation R_X86_64_32S against symbol `__libc_csu_fini' can not be used when making a PIE object; recompile with -fPIE
/home/dan/anaconda3/envs/sparta/bin/../lib/gcc/x86_64-conda-linux-gnu/12.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: failed to set dynamic section sizes: bad value
```

Has anyone else encountered this?

klingaard, 2024-03-05:
I don't have that issue, but a different one.  Specifically stf is built with LTO while the linking stage doesn't accept it.  Got a flag mismatch.

Have you tried with `clang`?

danbone, 2024-03-05:
Same error as this?

```
/home/dan/anaconda3/envs/riscv_perf_model/bin/../lib/gcc/x86_64-conda-linux-gnu/10.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: stf_lib/lib/libstf.a: error adding symbols: file format not recognized
clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/olympia.dir/build.make:131: olympia] Error 1
make[1]: *** [CMakeFiles/Makefile2:360: CMakeFiles/olympia.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

danbone, 2024-03-05:
Managed to fix it by moving the setup_stf_linker out of the Release build only

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index bee99e2..4fc4298 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -38,9 +38,10 @@ set (DISABLE_STF_DOXYGEN ON)

 if (CMAKE_BUILD_TYPE MATCHES "^[Rr]elease")
   set (FULL_LTO true)
-  include(${STF_LIB_BASE}/cmake/stf_linker_setup.cmake)
-  setup_stf_linker(false)
 endif()
+include(${STF_LIB_BASE}/cmake/stf_linker_setup.cmake)
+setup_stf_linker(false)
+

 # Use ccache if installed
 find_program (CCACHE_PROGRAM ccache)
```

Will create PR

klingaard, 2024-03-06:
Yep. That’s the one. Thanks for debugging

Shubhf, 2024-08-13:
@danbone Can I take up this issue?

klingaard, 2024-08-14:
Unless Dan disagrees, feel free to take it.  I just reproduced the original issue.

danbone, 2024-08-14:
@klingaard didn't this get fixed by
https://github.com/riscv-software-src/riscv-perf-model/commit/9d9a345060039594b2782520e743ab5ec2a613d3

Shubhf, 2024-08-14:
@danbone Can I take up this issue?
If you allow me to do so..
I think in this issue we are supposed to work on "-fPIE" error during compilation, with "CMAKE_BUILD_TYPE=Profile" in conda environment. It will be solved by adjusting compiler flags for PIE compatiability. I am interested to work on this

danbone, 2024-08-15:
Yes, absolutely fine by me