riscv-software-src/riscv-perf-model #169: dromajo SHA

jeffnye-gh, 2024-05-09:
We are going to make changes to our fork of dromajo.  I went to the Olympia read me on traces, to verify that a specific SHA for olympia/dromajo clone/build is specified rather than cloning HEAD.

I noticed the oly read-me page still references the previous patch mechanism. 

Is this the mechanism olympia is using rather than using the Condor fork?I don't have a preference, it is not my call. It is about making sure the docs are aligned.

If you are using the patch mechanism then I don't need to check for a SHA in your docs.

If you are using the CPM fork, then the READ-ME needs an update to specify the SHA from Kathlene's commit.

This readme:
https://github.com/riscv-software-src/riscv-perf-model/blob/master/traces/README.md

The current dromajo HEAD has KM's changes merged:
 ff13255a50bd1b5e7597f3bf2ceaf24b782e8b72

klingaard, 2024-05-09:
Ugh, I actually forgot to finish the "port" of the olympia over to using the fork. I'll try to find some time to finish that up.

jeffnye-gh, 2024-05-09:
no worries. 

for completeness this is the url to our fork:
https://github.com/Condor-Performance-Modeling/dromajo