riscv-software-src/riscv-perf-model #176: performance study: fusion

jeffnye-gh, 2024-06-18:
We propose a project to explore/compare the performance characteristics when instruction operations are fused in the machine alone and when the machine is assisted by a fusion aware compiler.

The project consists of two main components. The first component involves the more standard method of performing fusion within the machine, typically at the decode stage. Olympia supports fusion through the FSL API and domain language. As part of this project, we will continue this work, using the performance model to qualify instruction tuples and gauge their benefits.

The second component is a hardware/software co-design effort. This involves using fusion-aware compilers in combination with Olympia's fusion decoder. LLVM is known to support user-defined expressions of preferred instruction tuple sequences. 

The outcome would be a better understanding of the interaction of these two systems.

Shubhf, 2024-08-13:
@jeffnye-gh Can I take up this issue?

jeffnye-gh, 2024-08-13:
Absolutely. There are a few people interested in this, the more the merrier. 

The convention used in this repo has been to open a discussion and reference this issue, and then use the discussion thread to brainstorm and coordinate.

closed 2025-07-01 by jeffnye-gh