riscv-software-src/riscv-perf-model #185: Build problems

orion160, 2024-07-21:
Hi, I am starting on Olympia and having some problems with respect the build of various components which I already solved.

The last thing that I am doing is trying to generate a trace via Dromajo with drystone benchmark. But it seems that it needs a backward version of gcc/glibc.

Do you have a version which for the moment I can build Dromajo?

Thanks!

klingaard, 2024-07-21:
Hi Isaac, the Condor folks have a port of Dromajo that you might try: https://github.com/riscv-software-src/riscv-perf-model/issues/169#issuecomment-2103275637.  I've been meaning to update Olympia to point to this fork for some time, but haven't had an opportunity to do so (want to test it, etc).

Curious, what build issues did you incur while getting Olympia up?

orion160, 2024-07-21:
```bash
[  0%] Built target stf_git_version
[ 10%] Built target mavis
[ 16%] Built target mss
[ 18%] Building CXX object core/CMakeFiles/core.dir/FusionDecode.cpp.o
[ 20%] Building CXX object core/CMakeFiles/core.dir/Core.cpp.o
[ 22%] Building CXX object core/CMakeFiles/core.dir/SimpleBranchPred.cpp.o
[ 23%] Building CXX object core/CMakeFiles/core.dir/Fetch.cpp.o
[ 25%] Building CXX object core/CMakeFiles/core.dir/Decode.cpp.o
[ 27%] Building CXX object core/CMakeFiles/core.dir/VectorUopGenerator.cpp.o
[ 28%] Building CXX object core/CMakeFiles/core.dir/Rename.cpp.o
[ 30%] Building CXX object core/CMakeFiles/core.dir/Dispatch.cpp.o
[ 32%] Building CXX object core/CMakeFiles/core.dir/Dispatcher.cpp.o
[ 35%] Building CXX object core/CMakeFiles/core.dir/DCache.cpp.o
[ 35%] Building CXX object core/CMakeFiles/core.dir/Execute.cpp.o
[ 37%] Building CXX object core/CMakeFiles/core.dir/ExecutePipe.cpp.o
[ 38%] Building CXX object core/CMakeFiles/core.dir/Inst.cpp.o
[ 40%] Building CXX object core/CMakeFiles/core.dir/InstArchInfo.cpp.o
[ 44%] Building CXX object core/CMakeFiles/core.dir/LSU.cpp.o
[ 44%] Building CXX object core/CMakeFiles/core.dir/InstGenerator.cpp.o
[ 45%] Building CXX object core/CMakeFiles/core.dir/InstGroup.cpp.o
[ 47%] Building CXX object core/CMakeFiles/core.dir/IssueQueue.cpp.o
[ 50%] Building CXX object core/CMakeFiles/core.dir/Preloader.cpp.o
[ 52%] Building CXX object core/CMakeFiles/core.dir/MMU.cpp.o
[ 54%] Building CXX object core/CMakeFiles/core.dir/ROB.cpp.o
[ 54%] Building CXX object core/CMakeFiles/core.dir/CPU.cpp.o
[ 55%] Building CXX object core/CMakeFiles/core.dir/MavisUnit.cpp.o
[ 59%] Building CXX object core/CMakeFiles/core.dir/CPUTopology.cpp.o
[ 59%] Building CXX object core/CMakeFiles/core.dir/CPUFactory.cpp.o
[ 61%] Building CXX object stf_lib/lib/CMakeFiles/stf.dir/stf_protocol_types.cpp.o
In file included from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_ifstream.hpp:14,
                 from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_factory.hpp:11,
                 from /home/cc/riscv-perf-model/stf_lib/lib/stf_protocol_types.cpp:1:
/home/cc/riscv-perf-model/stf_lib/stf-inc/stf_protocol_fields.hpp: In instantiation of 'void stf::defaultProtocolFieldFormatter(std::ostream&, const char*, const Value&) [with Value = stf::SerializableVector<unsigned char, short unsigned int>; std::ostream = std::basic_ostream<char>]':
/home/cc/riscv-perf-model/stf_lib/stf-inc/protocols/tilelink.hpp:27:13:   required from here
/home/cc/riscv-perf-model/stf_lib/stf-inc/stf_protocol_fields.hpp:261:32: error: no matching function for call to 'stf::format_utils::formatHex(std::ostream&, const stf::SerializableVector<unsigned char, short unsigned int>&)'
  261 |         format_utils::formatHex(os, data);
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_protocol_fields.hpp:14,
                 from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_ifstream.hpp:14,
                 from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_factory.hpp:11,
                 from /home/cc/riscv-perf-model/stf_lib/lib/stf_protocol_types.cpp:1:
/home/cc/riscv-perf-model/stf_lib/stf-inc/format_utils.hpp:232:13: note: candidate: 'template<class OStream, class T> static std::enable_if_t<is_integral_v<T> > stf::format_utils::formatHex(OStream&&, T, int, char)'
  232 |             formatHex(OStream&& os, const T val, const int width = numHexDigits<T>(), const char pad_char = '0') {
      |             ^~~~~~~~~
/home/cc/riscv-perf-model/stf_lib/stf-inc/format_utils.hpp:232:13: note:   template argument deduction/substitution failed:
In file included from /home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/bits/move.h:57,
                 from /home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/bits/stl_pair.h:59,
                 from /home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/utility:70,
                 from /home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/array:38,
                 from /home/cc/riscv-perf-model/stf_lib/stf-inc/stf_factory.hpp:4,
                 from /home/cc/riscv-perf-model/stf_lib/lib/stf_protocol_types.cpp:1:
/home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/type_traits: In substitution of 'template<bool _Cond, class _Tp> using enable_if_t = typename std::enable_if::type [with bool _Cond = false; _Tp = void]':
/home/cc/riscv-perf-model/stf_lib/stf-inc/format_utils.hpp:232:13:   required by substitution of 'template<class OStream, class T> static std::enable_if_t<is_integral_v<T> > stf::format_utils::formatHex(OStream&&, T, int, char) [with OStream = std::basic_ostream<char>&; T = stf::SerializableVector<unsigned char, short unsigned int>]'
/home/cc/riscv-perf-model/stf_lib/stf-inc/stf_protocol_fields.hpp:261:32:   required from 'void stf::defaultProtocolFieldFormatter(std::ostream&, const char*, const Value&) [with Value = stf::SerializableVector<unsigned char, short unsigned int>; std::ostream = std::basic_ostream<char>]'
/home/cc/riscv-perf-model/stf_lib/stf-inc/protocols/tilelink.hpp:27:13:   required from here
/home/cc/miniconda3/envs/riscv_perf_model/x86_64-conda-linux-gnu/include/c++/10.4.0/type_traits:2554:11: error: no type named 'type' in 'struct std::enable_if<false, void>'
[... cut ...]
cc1plus: note: unrecognized command-line option '-Wno-gnu-zero-variadic-macro-arguments' may have been intended to silence earlier diagnostics
make[2]: *** [stf_lib/lib/CMakeFiles/stf.dir/build.make:174: stf_lib/lib/CMakeFiles/stf.dir/stf_protocol_types.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:682: stf_lib/lib/CMakeFiles/stf.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 62%] Linking CXX static library libcore.a
[ 62%] Built target core
make: *** [Makefile:91: all] Error 2
```

This is the error when I use the conda environment.

Managing the dependencies manually and with the distro gcc version it works.

orion160, 2024-07-21:
@klingaard, thank you. I'll checkout the condor fork

klingaard, 2024-07-21:
Filed an issue with STF lib: https://github.com/sparcians/stf_lib/issues/39

klingaard, 2024-07-21:
Might need to know the compiler version, etc if you can supply that.

jeffnye-gh, 2024-07-22:
re: miniconda version

We use this version in production: py311_23.9.0-0

We are testing this version: py312_24.3.0-0, this seems to work, but we have not moved into main yet.

We ran into trouble with these:
"Issues testing against Conda 24.5.0 Python 3.12.4 released Jun 26, 2024, I’ll also test against older version (24.3)."

Could be specific to us or could be a general issue.

PR #188 (2024-07-23): Update to latest stf_lib
Fixes #185
commit 403f9ac8 (2024-07-23): Update to latest stf_lib (#188)

Fixes #185
closed 2024-07-23 by AaronGChan