riscv-software-src/riscv-perf-model #189: FSL Build Issue

aarongchan, 2024-07-31:
Running into this issue after merging in the most recent FSL changes into my local branch, trying to build Olympia to test VLSU changes but this is blocking my build from completing:
```
src/fsl.l:77:9: error: 'RET' macro redefined [-Werror,-Wmacro-redefined]
#define RET(TOK) return TOK
        ^
/Users/aaronchan/Documents/riscv/riscv-perf-model/release/fsl/fsl_interp/yy.tab.hpp:152:9: note: previous definition is here
#define RET 288
        ^
/Users/aaronchan/Documents/riscv/riscv-perf-model/release/fsl/fsl_interp/lex.yy.cpp:830:9: error: 'SEQUENCE' macro redefined [-Werror,-Wmacro-redefined]
#define SEQUENCE 2
        ^
/Users/aaronchan/Documents/riscv/riscv-perf-model/release/fsl/fsl_interp/yy.tab.hpp:155:9: note: previous definition is here
#define SEQUENCE 291
        ^
/Users/aaronchan/Documents/riscv/riscv-perf-model/release/fsl/fsl_interp/lex.yy.cpp:1929:38: error: comparison of integers of different signs: 'unsigned long' and 'int' [-Werror,-Wsign-compare]
        if (((yy_n_chars) + number_to_move) > YY_CURRENT_BUFFER_LVALUE->yy_buf_size) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

jeffnye-gh, 2024-08-13:
I am very sorry. I have been out of contact longer than expected. I just saw this today. 

I have submitted a PR, [#199](https://github.com/riscv-software-src/riscv-perf-model/pull/199)

I am not able to duplicate the problem on my system but I suspect this to be a Flex/Bison version issue. The version requirements were not in CMakelists, they should have been.

I've also added some guards to the macros in the flex file. 
And I have added Wall/Werror to cmake.

This warning/error:
/Users/aaronchan/Documents/riscv/riscv-perf-model/release/fsl/fsl_interp/lex.yy.cpp:830:9: error: 'SEQUENCE' macro redefined [-Werror,-Wmacro-redefined]
#define SEQUENCE 2

..comes from a Flex generated file. I believe using the versions of Flex in cmakelists will remove the warning/error.

I apologize for the late response.

jeffnye-gh, 2024-08-21:
This fix has been merged.

closed 2024-08-21 by aarongchan