riscv-software-src/riscv-perf-model #193: Use HTTPS instead of SSH for submodules?

oms-vmicro, 2024-08-05:
There have been recent changes to the method `riscv-perf-model` clones its submodules.  Currently it is mixed with `stf_lib` and `fsl` using HTTPS and `mavis` using SSH ([link](https://github.com/riscv-software-src/riscv-perf-model/blob/master/.gitmodules)).

I'd like to propose `mavis` also switch back to using HTTPS so all submodules are cloned in a consistent way.  Though I'd like to understand if there's a reason to use SSH over HTTPS and/or do so for `mavis` but not `stf_lib` and `fsl`.

My basis for proposing HTTPS is that It is possible not all consumers of the `riscv-perf-model` will want to use SSH as I believe that requires them to have a GitHub account.  For read-only consumers, HTTPS is sufficient.  Non-GitHub users can still get around this by modifying their `.gitmodules` to use HTTPS instead of SSH but it's another step in getting started.

@jeffnye-gh @kathlenemagnus

Shubhf, 2024-08-08:
hey @oms-vmicro Can I take up this issue?
I am thinking that switching mavis  to HTTPS would be a better option , as it would be a consistent and straightforward access for all users, particularly without git accounts, This not only helps to making it easier for new contributors, and indirectly will be helping new contributors to work with risk-perf-model

klingaard, 2024-08-08:
There's no strong need.  Feel free to change it.

Shubhf, 2024-08-10:
I have tried to make necessary changes @klingaard @oms-vmicro  Pls review and suggest what all I could do more[https://github.com/riscv-software-src/riscv-perf-model/pull/195
](#195 )

closed 2024-08-15 by klingaard