riscv-software-src/riscv-perf-model #197: Hello all!

Komal0103, 2024-08-11:


klingaard, 2024-08-14:
Those are new issues

Looks like you're running directly from the build (as per the tutorial).  This _should_ be ok, but TBH I haven't tested that in a long while.  We moved to an installed version of argos in the main repo.  I'll look into it.

Can you try installing the pipe viewer?  In other words, try this:
```
cmake --install . --prefix=$CONDA_PREFIX
``` 
then try this:
```
argos -d small_core_pipeout -l ../layouts/small_core.alf
```

Komal0103, 2024-08-15:
Thank you for your reply! I tried using this. I get the same error:

````
Traceback (most recent call last):
  File "/home/komals/miniconda3/envs/sparta/bin/argos", line 5, in <module>
    from pipe_view.argos import main
  File "/home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/argos.py", line 41, in <module>
    from pipe_view.gui.dialogs.select_layout_dlg import SelectLayoutDlg  # noqa: E402, E501
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/gui/dialogs/select_layout_dlg.py", line 9, in <module>
    from ...model.layout import Layout
  File "/home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/model/layout.py", line 11, in <module>
    from .import element_types as etypes
  File "/home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/model/element_types.py", line 2, in <module>
    from .element import FakeElement, BoxElement, ImageElement
  File "/home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/model/element.py", line 30, in <module>
    from ..logsearch import LogSearch  # Argos module for searching logfiles
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: /home/komals/miniconda3/envs/sparta/lib/python3.12/site-packages/pipe_view/logsearch.cpython-312-x86_64-linux-gnu.so: undefined symbol: _ZTI8wxThread
````

This is the same I get while using directly from the build.

klingaard, 2024-08-15:
What linux are you using?

Komal0103, 2024-08-15:
I am using Ubuntu 20.04.

klingaard, 2024-08-17:
Ok, it used to work fine on 20, but I’m using 22.  Anyway, I assume you’re using a conda environment?  Also what compiler / version are you using?