riscv-software-src/riscv-perf-model #255: Add Support for Vector Loads and Stores

kathlenemagnus-mips, 2025-05-16:
Extended Olympia to support vector load and store instructions.

Extend the current LSU so that it can "unroll" a vector instruction and generate separate memory accesses for each active vector element. For simplicity, read and write ports to the VRF (vector register file) can be added to the LSU for writing load data and reading store data, the vector mask register and the index values (for indexed loads and stores).

The design should include the following components:
- An "address unroller" for generating load/store requests from a vector instruction.
- A buffer for holding load data before writing the data into the VRF

Resources:
[The RISC-V Vector ISA Tutorial](https://riscv.org/wp-content/uploads/2024/12/15.20-15.55-18.05.06.VEXT-bcn-v1.pdf) (good intro, but it's for RVV 0.7 so beware!)
[RISC-V Vector in a Nutshell](https://fprox.substack.com/p/risc-v-vector-in-a-nutshell)
[Ara2: Exploring Single- and Multi-Core Vector Processing with an Efficient RVV 1.0 Compliant Open-Source Processor](https://arxiv.org/abs/2311.07493) (inspiration?)

aarongchan, 2025-05-18:
[This is a presentation video with the RISC-V ISA Tutorial slides you linked that I used to learn the ISA](https://www.youtube.com/watch?v=GzZ-8bHsD5s)
[Another good presentation on RISC-V Vector ISA](https://eupilot.eu/wp-content/uploads/2022/11/RISC-V-VectorExtension-1-1.pdf)

This is probably getting to far ahead of basic vector support in this model, but I know a a global VRF vs banked VRF was a design proposal that I read about when I was learning about RISC-V vector. I see in the Ara2 paper that they show each lane having a VRF chunk. So it would be cool if the work that gets done around adding VLSU can keep that in mind during the implementation.

htanna-ventanamicro, 2025-07-02:
Hi @kathlenemagnus-mips and @aarongchan is there anything in that is already implemented here? (Maybe in personal repo but is not pushed here). How much of this is pending and would also like to scope it out as I want to contribute to this issue. Please lmk.

aarongchan, 2025-07-07:
This is the PR for my VLSU implementation on the older LSU pipeline:
https://github.com/riscv-software-src/riscv-perf-model/pull/219

2elliti, 2026-10-05:
hey can i have the ticket? If this is feature request is till open?