riscv-software-src/riscv-perf-model #270: Compilation failure during Embench benchmark build process when targeting QEMU-based execution.

aditi75432, 2025-07-28:
# Build fails for bare-metal RISC-V target with "Makefile not found" error in Docker

## Summary

I am attempting to compile the Embench-IoT suite for a bare-metal RISC-V target (`rv32`) inside a Docker container. The build process consistently fails at the `make` step, reporting that no Makefile can be found, even though the repository is cloned recursively.

## Steps to Reproduce
Kindly follow this repo-
https://github.com/aditi75432/qemu-benchmarks/tree/main/embench

## Actual Behavior

During the Docker build process, the following error occurs:

```bash
---> Running in <container_id>
make: *** No targets specified and no makefile found. Stop.
The command '/bin/sh -c make PORT=baremetal TARGET=rv32' returned a non-zero code: 2

aditi75432, 2025-07-31:
> Please point me to a) steps to set up docker followed by b) cloning embench, c) the specific command being invoked to build embench and d) working directory from which the command is being invoked.

Sir kindly refer to this repo - https://github.com/aditi75432/qemu-benchmarks/blob/main/embench/Instructions.md , it has all the documents (I separated the docker for embench for now so that we could identify the core issue)

vinicius-r-silva, 2025-08-01:
It is possible to compile using the following command:

```bash
python3 ./build_all.py --arch riscv32 --chip generic --board ri5cyverilator \
  --cc riscv32-unknown-elf-gcc \
  --cflags="-c -O2 -ffunction-sections -march=rv32imfdc -mabi=ilp32d" \
  --ldflags="-Wl,-gc-sections" \
  --user-libs="-lm"
```

For this to work, I needed to have the RISC-V toolchain compiled with:

```bash
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make
```

It may also be possible to use the `--enable-multilib` option instead of `--with-arch`, but i didn't test it.

Additionally, it's necessary to **comment out line 36** in `src/wikisort/libwikisort.c`:

```c
// typedef uint8_t bool;
```

Without this change, the compiler throws the following error:

```
/embench-iot/src/wikisort/libwikisort.c:36:17: error: two or more data types in declaration specifiers
   36 | typedef uint8_t bool;
```

A more elegant solution might involve using predefined compiler macros (e.g., checking if `bool` is already defined), but for now, simply commenting out the line is sufficient.

vinicius-r-silva, 2025-08-07:
working with @aditi75432, the following Dockerfile was created to build embench
```
FROM ubuntu:22.04

RUN apt-get update
RUN DEBIAN_FRONTEND=noninteractive TZ=Etc/UTC apt-get install -y device-tree-compiler libboost-regex-dev libboost-system-dev git binutils build-essential cmake gcc-riscv64-linux-gnu zstd libzstd-dev autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev libslirp-dev git-email libaio-dev libbluetooth-dev libcapstone-dev libbrlapi-dev libbz2-dev libcap-ng-dev libcurl4-gnutls-dev libgtk-3-dev libibverbs-dev libjpeg8-dev libncurses5-dev libnuma-dev librbd-dev librdmacm-dev libsasl2-dev libsdl2-dev libseccomp-dev libsnappy-dev libssh-dev libvde-dev libvdeplug-dev libvte-2.91-dev libxen-dev liblzo2-dev valgrind xfslibs-dev libnfs-dev libiscsi-dev gcc binutils python3-pip python3-sphinx python3-sphinx-rtd-theme ninja-build flex bison git autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential texinfo gperf libtool patchutils python3-tomli bc zlib1g-dev libexpat-dev cmake libglib2.0-dev gcc-riscv64-linux-gnu libslirp-dev 

RUN git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
RUN git clone --recursive https://github.com/embench/embench-iot.git
RUN git clone https://gitlab.com/qemu-project/qemu.git

WORKDIR /qemu/build
RUN ../configure --target-list=riscv32-linux-user,riscv64-linux-user,riscv32-softmmu,riscv64-softmmu
RUN make -j$(nproc)
RUN make install
ENV QEMU_PLUGINS=/qemu/build/contrib/plugins

WORKDIR /riscv-gnu-toolchain
RUN ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
ENV PATH=/opt/riscv/bin:$PATH
RUN make -j$(nproc)

WORKDIR /
RUN sed -i '36s/^typedef uint8_t bool;/\/\/ &/' /embench-iot/src/wikisort/libwikisort.c

WORKDIR /embench-iot
RUN python3 ./build_all.py --arch riscv32 --chip generic --board ri5cyverilator \
    --cc riscv32-unknown-elf-gcc \
    --cflags="-c -O2 -ffunction-sections -march=rv32imfdc -mabi=ilp32d" \
    --ldflags="-Wl,-gc-sections" \
    --user-libs="-lm"
``` 

It is possible to generate BBVs from one of embench workloads with the commands:
```
$ docker buildx build embench .
$ docker run --rm embench \
  bash -c "
    qemu-riscv32 -plugin /qemu/build/contrib/plugins/libbbv.so,interval=100000,outfile=/embench-iot/bd/src/crc32/crc32.bb /embench-iot/bd/src/crc32/crc32 && cat /embench-iot/bd/src/crc32/crc32.bb.0.bb
  "
```

There is still some work to be done to integrate this to the existing Dockerfiles:
* Compiling the riscv toolchain to also be able to compile 64 bits
* Create some way to easily execute any of the embenchs workloads (e.g. move workload binaries to a single folder, create helper scripts, etc.)

Some others possible improvements:
* Replace the `sed` workaround in the dockerfile to something more cleaner / robust (maybe fork Embench with the corrections, or open an issue in their repository?)
* Test compiling the riscv toolchain with multilib, instead of having separated compilers for 32 and 64 bits

All things aside, the issue of having to compile embench seems to resolved on my view

aditi75432, 2025-08-10:
### Issue Resolution: Switched to a Python-based build and QEMU User Emulation

Successfully resolved this issue with @vinicius-r-silva . The initial "Makefile not found" error was a result of trying to use a `make`-based approach for a bare-metal target, which was not the intended build method for this configuration.

The solution was to pivot to a more robust and automated approach using Docker, the official Embench build script, and QEMU's user-mode emulation.

#### Key Changes in the Solution:

1.  **Build System:** Instead of `make`, I am now using the project's own `build_all.py` script. This script correctly handles the configuration and compilation for the `riscv32` target.

2.  **Pre-Built Toolchain:** To significantly speed up the build process and ensure stability, the Dockerfile now uses the `ribeirovsilva/riscv-toolchain:latest` image. This provides a ready-to-use RISC-V GCC toolchain without needing to build it from source.

3.  **Simplified QEMU Integration:** Instead of a complex build of QEMU from source, the environment now installs `qemu-user` and `qemu-user-static` via `apt-get`. This is a much simpler and more reliable method for enabling the execution of cross-compiled RISC-V binaries on a standard host.

4.  **Automation Scripts:** The entire process is now managed by two new scripts:
    * `simple_setup.sh`: An orchestrator script on the host machine to build the Docker image and run benchmarks.
    * `run_embench_simple.sh`: An executor script inside the container that runs the specific workloads using `qemu-riscv32`.

#### Verification of Success

The new setup successfully compiles all 22 Embench workloads and executes them within the container, as shown by the demo output:

```shell
Running Embench demonstration...

Step 1: Listing available workloads...
================================
... (list of 22 workloads) ...
================================
Total: 22 workloads

Step 2: Running crc32 workload with timing...
Successfully ran crc32

Step 3: Running a few more workloads...
Successfully ran md5sum
Successfully ran cubic
Successfully ran nbody

Demo completed successfully!
```

Please refer - https://github.com/aditi75432/qemu-benchmarks/tree/main/EMBENCH

closed 2025-08-10 by aditi75432